fifonik/ElegooNeptuneThumbnailPrusaMod #1: Adding Cura_SteamEngine breaks PrusaSlicer's g-code preview 'Features' functionality

fifonik, 2023-10-13:
Looks like when 'Cura' is present in g-code, PrusaSlicer's g-code viewer expect `;TYPE:...` comments contain different values.

Prusa:
External perimeter      Overhang perimeter      Internal infill Bridge infill   Skirt/Brim      Wipe tower

Cura:
SKIRT   WALL-OUTER      WALL-INNER  SKIN    FILL        SUPPORT SUPPORT-INTERFACE       PRIME-TOWERBRIDGE

Felix14-v2, 2025-02-05:
Also Orca refuses to open processed gcodes with an error "generated not in OrcaSlicer", so OrcaSlicer is affected too

fifonik, 2025-02-05:
This is expected behaviour as the script is masking the fact that the gcode was generated by OrcaSlicer.

If you use the script on N4 series with modern firmware, you can try to use option `--modify_slicer_header=false` to let script know that you do not want it to do that.
Unfortunately, I did not have time to test it yet by myself so I do not know if thumbnail will be displayed on printer's screen in this case or not (this is the reason I have not added the option on the script's main github page).