fifonik/ElegooNeptuneThumbnailPrusaMod #5: Execution of thumbnail executable results in DLL reference failure

paul95831, 2023-12-30:
Host system, Windows 10
Orca Slicer 1.8.1

Running the .exe executable either directly from File Manager or while slicing in Orca Slicer results in the following windows popup error:

X Failed to execute script 'thumbnail' due to unhandled exception: DLL load failed while importing _ctypes: The specified module could not be found.

Traceback (most recent call last):
  File "J:\work\programming\Python\ElegooNeptuneThumbnailPrusaMod\ElegooNeptuneThumbnailPrusaMod\thumbnail.py", line 16, in <module>
  File "PyInstaller\loader\pyimod02_importers.py", line 391, in exec_module
  File "ctypes\__init__.py", line 8, in <module>
ImportError: DLL load failed while importing _ctypes: The specified module could not be found.

Should an additional DLL file be included and placed alongside the .exe file?

fifonik, 2024-01-30:
Hm. 
I thought everything should be included in compiled executable. 
But I'll double check. Thanks for the info.

fifonik, 2024-01-30:
If you use `thumbnail.exe` you do not need the dll to be aside as it is integrated into the executable.
The bundled executable extracted into separate folder in system %temp% and executed from there.
This is what I see in thumbnail.log when using the bundled executable located at "c:\Temp\ElegooNeptuneThumbnail\thumbnail.exe":
2024-01-30 17:26:48,207 - INFO - Input file: C:\Users\[USER]\AppData\Local\Temp/orcaslicer_model/Tue_Jan_30/17_24_32#4888#49/Metadata\.4888.0.gcode.pp
2024-01-30 17:26:48,207 - INFO - The first thumbnail from input file larger than 100x100 will be used
2024-01-30 17:26:48,207 - INFO - Using short pring duration format
2024-01-30 17:26:48,207 - INFO - Original inage will be updated
2024-01-30 17:26:48,207 - DEBUG - Parsing file
...
2024-01-30 17:26:48,699 - DEBUG - Encoding image for new printers (;gimage:)
2024-01-30 17:26:48,768 - DEBUG - Using Windows dll: C:\Users\[USER]\AppData\Local\Temp\_MEI18802\ColPic_X64.dll
2024-01-30 17:26:48,812 - DEBUG - Scaling image to 160x160
...
2024-01-30 17:26:48,846 - DEBUG - Writing new header with image into file C:\Users\[USER]\AppData\Local\Temp/orcaslicer_model/Tue_Jan_30/17_24_32#4888#49/Metadata\.4888.0.gcode.pp.output
2024-01-30 17:26:48,846 - DEBUG - Copying content from file C:\Users\[USER]\AppData\Local\Temp/orcaslicer_model/Tue_Jan_30/17_24_32#4888#49/Metadata\.4888.0.gcode.pp to file C:\Users\[USER]\AppData\Local\Temp/orcaslicer_model/Tue_Jan_30/17_24_32#4888#49/Metadata\.4888.0.gcode.pp.output

If you use `thumbnail.py` then you need to put the dll aside (I'll adjust readme).
Thanks.

fifonik, 2024-01-30:
If you use exe and still have the issue -- you should check if anti-virus program is not interfere with the process.

paul95831, 2024-01-30:
Certainly something unique to my computer.  1, other people would report this if it were happening elsewhere, and 2, I just ran this on my friends computer with no issues (fresh Windows install about two months ago).  My tests have been with and without anti virus, and with admin privileges (cmd prompt) with it failing the same way.  You may close this open ticket and with shovel in hand, I go a dig'n.

fifonik, 2024-01-31:
1. You can try to add option "--debug", may be you will notice something strange in log (for example, system %temp% pointing to non-existing or read-only location)
2. As this is exception, the temp folder have to be kept in your system somewhere. Try to find 'thumbnail.py' file and then check if the dll is in the same folder.

paul95831, 2024-02-14:
This was ultimately down to a Windows problem but unsure exactly what and where.  Tracked all file and registry operations and compared with another Windows system that it ran fine on and still unable to overcome the DLL failure.  Rebuilt the Windows 10 system and all is fine now.

closed 2024-02-14 by paul95831