fifonik/ElegooNeptuneThumbnailPrusaMod #8: SyntaxError - Ubuntu

tomgit33, 2024-02-28:
Ubuntu python error:

<title>ElegooNeptuneThumbnailPrusaMod/thumbnail.py at main · fifonik/ElegooNeptuneThumbnailPrusaMod · GitHub</title>
SyntaxError: invalid character '·' (U+00B7)

How to solve?

fifonik, 2025-01-21:
Sorry, email notifications were turned off for the repository so I have not seen all the issues :(

Hm. 
I'm using unicode characters in script and looks like your linux does not like them.
I have not had such issues on my Ubuntu based Mint, but it is possible that I have some fonts installed.
I will try to investigate how to add unicode characters as entities in python.

fifonik, 2025-01-21:
Wait a moment, there is no such character ('·' with code U+00B7) in the thumbnail.py

fifonik, 2025-03-18:
Closing as origin unknown and no further feedback from user

closed 2025-03-18 by fifonik