fifonik/ElegooNeptuneThumbnailPrusaMod #10: Orca cannot find a file when there are blank space in the path

haski369, 2024-05-03:
[image]
Changing it from a path that has blanks in the name of the path to the main directory with zero "spacebars" added Fixes it.
[image]

haski369, 2024-05-03:
Also, after adding thumbnail mod i got this error, just First flowrate test 
[image]

fifonik, 2025-01-21:
> Orca cannot find a file when there are blank space

As per installation steps, you have to enclose path into quotes in this case.

fifonik, 2025-01-21:
> Also, after adding thumbnail mod i got this error

If you see such error this means that script executed successfully and then ur slicer sent the processed gcode file to printer.
This is the error returned by fluidd.
In the error message I see: **File currently in use**
I'm having such issue if I send file to printer, start printing and then trying to send file with the same name again while printer is still printing it. This is not related to the thumbnail generation script.

closed 2025-01-21 by fifonik