fifonik/ElegooNeptuneThumbnailPrusaMod #11: Missing additional print information

CLNilix, 2024-05-28:
Hi,
i am just trying your mod on orca slicer 2.0.0 and noticed, that the additional print information is missing on the thumbnail:
[image]

Did you get this information from the gcode-file?
then i think the version of orca slicer i use, did not write this information in the gcode-file.
Any idea which setting in orca is responsible for this information?

CLNilix, 2024-05-28:
UPDATE
after some research, it looks like the needed information is included in the gcode file from orca-slicer (and prusa-slicer):
[image]

I assume your script did not interpret this information right.
i tried the following slicers and just got the thumbnail without any additional print information:

- Orca Slicer 2.0.0
- Prusa Slicer 2.6.1+win64

maybe it would be a better idea to transform the META data from orca/prusa to the cura format.
I get the thumbnail and the additional print data (see second print file in the first post) on my Neptune 4 with the following format of Cura 5.7.0:
[image]

CLNilix, 2024-05-29:
UPDATE
i startet the print and the thumbnail on the **printer LCD** contains the additional print information.
It seems only the thumbnail in Fluidd did not contain the additional print information.
Any idea how to archive that?

fifonik, 2025-01-21:
If I understand correctly, this is duplicate for #1
Elegoo printers not displaying thumbnail if NOT Cura mentioned in gcode. So the script is adding Cura's header and masking Orca/Prusa headers to bypass this.
However, the script is not replacing anything else in gcode file and other software (as Prusa/Orca gcode viewer or fluidd backend) is failing as they expecting details in Cura's format :(

I do not know an easy way to fix that (converting everything from Orca/Prusa format to Cura format is complicated).
It is possible that in newer firmwate Elegoo removed that restriction and now the masking is not required.
I'm planning to check it out in nearest future.

fifonik, 2025-02-04:
If you use the script with N4 series and have the most recent firmware installed on your printer, you can try new option `--modify_slicer_header=false` that will turn off the slicer header modification.
It is possible that this will fix the additional information in fluidd.

fifonik, 2025-02-04:
> UPDATE i startet the print and the thumbnail on the **printer LCD** contains the additional print information. It seems only the thumbnail in Fluidd did not contain the additional print information. Any idea how to archive that?

By default the script is not modifying original image.
Just use an option `--update_original_image` to change this.

TrapStoner, 2025-03-13:
In Orca 2.3.0-rc I get error script failed with error code 2 when i try to use `--modify_slicer_header=false` , without it everything works. Here is my full path `"C:\thumbnail.exe" "--short_duration_format" "--update_original_image" "--modify_slicer_header=false"` am i missing something? Im running openneptune so i don't need cura header spoofing and want to retain orcaslicer header info.

fifonik, 2025-03-13:
Oh, I have not checked how the false argument parsing =)
It turned out that Pythod is unique here as well.
Well, I've pushed changed version where the followin option can be used to turn off the slicer header modification:
`--no-modify_slicer_header`

TrapStoner, 2025-03-13:
Amazing! Thank you very much it works now! Maybe this issue can be closed and ` --no-modify_slicer_header` can be properly documented in the readme now.

fifonik, 2025-03-18:
What is your printer and what FW?
My Neptune 4 still has old firmware so do not show thumbnail with the option --no-modify_slicer_header.
It would be nice to have some feedback from users who using the option with new FW:
- Does it show preview on printer screen?
- Does fluidd show correct information in history?

mcelotto72, 2025-03-22:
Hi, I have a Neptune 3 Pro and I'm trying this with the new Elegoo Slicer (based on Orca)..  Just received this error  If I take the "--info" away then it works.  I've tried different variations of the "--info" and I can't seem to figure it out.  Maybe you could put an example i the readme?

```
Post-processing script "\\drobo\Drobo\3D Print
Files\Michael\ElegooSlicer_Windows_V1.1.8.2_portable\ElegooThumbnail\thumbnail.exe" "--info"
on file
C:\Users\mcelotto\AppData\Local\Temp/elegooslicer_model/Sat_Mar_22/17_39_33#46872#49/M
etadata\.46872.0.gcode.pp failed.

Error code: 2
```

fifonik, 2025-03-22:
You cannot just add `--info`.
It should be used as `"--info" "print_duration, model_height, filament_weight, filament_length"`
In the example (this is default) you will have:
- "print_duration" -- at top left corner
- "model_height"  -- at top right corner
- "filament_weight" -- at bottom left corner
- "filament_length" -- at bottom right corner

You can change the order of these items and/or use other values ("filament_cost" is also supported at the time of writing) to fit your needs.

P.S. You should not hijack into discussion that is not related to your issue. Create a new ticket please.

mcelotto72, 2025-03-22:
Thanks, sorry wasn't trying to hijack anything.  I'll make a new ticket, even with the example same error.

bckyrdbshr, 2025-05-09:
Sorry if im hijacking the thread, please forgive me. I'm new to all this stuff. I have an Elegoo N3 Plus I've been trying my best to get to work. Depending on firmware on my machine i might get a pic i may get a black screen. All in all when it is working correctly just get the picture, no other info? Is this correct. Also ive only been able to get this to work on the Elegoo version of Orca, when I try on the true Orca slicer I've yet to get it to work, says dir. doesnt exist or something similar to that. Basically no matter where i put the file and point it to the printer it never works. I dont know if im putting it in the wrong dir. or naming the printer wrong or what. It would be nice to have, Ive tried Cura and dont care for it. Any help would be greatly appreciated and please accept my apologies if I'm out of line. I'm sorry.

fifonik, 2025-05-09:
Could you please create new issue, put the above text there and add screenshots from slicer with thumbnail settings ("Printer Settings" | "Basic Information") and from printing settings ("Others" | "Post-Processing Scripts")
We will continue from there.
Thanks.