fifonik/ElegooNeptuneThumbnailPrusaMod #12: failed to run post-processing in Orca Slicer on Mac

cwiegert, 2024-06-06:
First Off... thanks for posting this utility, have been looking for something like this for a number of months.   Straight forward install and instructions - thanks for that as well. 
Installed the PyQt6, but doesn't seem to help with this error
```
Post-processing script "/usr/local/bin/python3"
"/usr/local/bin/ElegooNeptuneThumbnailPrusaMod-main/dist/thumbnail" "--image_size" "300x300" on
file /var/folders/3v/qd8pyr2j77n472dp0_n5tq7w0000gn/T/.OrcaSlicer.upload.969d-f8a9-8ef8-13e4
failed.
Error code: 1
Output:
SyntaxError: Non-UTF-8 code starting with '\xcf' in file
/usr/local/bin/ElegooNeptuneThumbnailPrusaMod-main/dist/thumbnail on line 1, but no encoding
declared; see https://peps.python.org/pep-0263/ for details
```

no matter what I try with paths and parameters, I get the same error.
here is the exact code from the Post-processing Scripts section

```"/usr/local/bin/python3" "/usr/local/bin/ElegooNeptuneThumbnailPrusaMod-main/dist/thumbnail" "--image_size" "300x300"```

cwiegert, 2024-06-06:
OK - I have it working.... running the script instead of the executable.   Also - I had to be careful about where to install the PYQT6, and make sure to put it in the appropriate version of python working directory.    using pip3.12 install instead of simply pip3 install.   by defualt the pip3 ran the install into site-packages for the 3.11 install of python.

Don't know what the issue is with the exe, but I have it working.

fifonik, 2025-01-21:
Well, the exe is supposed to be used on Windows.

fifonik, 2025-01-21:
If I understand correctly, this is duplicate for #1
Elegoo printers not displaying thumbnail if NOT Cura mentioned in gcode. So the script is adding Cura's header and masking Orca/Prusa headers to bypass this.
However, the script is not replacing anything else in gcode file and other software (as Prusa/Orca gcode viewer or fluidd backend) is failing as they expecting details in Cura's format :(

I do not know an easy way to fix that (converting everything from Orca/Prusa format to Cura format is complicated).
It is possible that in newer firmwate Elegoo removed that restriction and now the masking is not required.
I'm planning to check it out in nearest future.

closed 2025-03-18 by fifonik