fifonik/ElegooNeptuneThumbnailPrusaMod #15: Minimum thumbnail size 100x100 - not able to add 85x85 needed for some printers

Marooned-MB, 2024-11-20:
I can see that code discards input size less than 100x100px.

I'm using this for Artillery Sidewinder X3 Pro with Orca Slicer and in order to see thumbnails on the list of files, they must be 85x85px so this plugin can't generate them (without code modification and building locally).

Was there some rational reason behind such restriction and if not, could this be removed to allow for smaller thumbnails?

Thanks

Marooned-MB, 2024-11-20:
More info:
I've edited the file and build the project and now it accepts smaller resolutions but they are seen on the detail page but still not on the file list.

I'm trying to compare the output with working [Cura plugin](https://github.com/Molodos/ElegooNeptuneThumbnails) - there are a lot of differences. Maybe I could combine those two plugins into one working. Will let you know if that happens.

Marooned-MB, 2024-11-20:
Got it.
Changing lines [452..453](https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod/blob/main/thumbnail.py#L452) into
```Python
            header += self.image_encode_new(self.image_modify(self.image_resize(img, 85, 85)), ';simage:')
            header += self.image_encode_new(self.image_modify(self.image_resize(img, 230, 230)), ';gimage:')
            header += self.image_encode_new(self.image_modify(self.image_resize(img, 170, 170)), ';mimage:')
```
gave me the working results. No need to change accepting smaller thumbnails as stated initially as I understood the code that it looks embedded thumbnails and just get the 1st bigger than 100x100 so for my case it's the best to set Orca to generate one thumbnail 230x230/PNG and it will be used for all the three mentioned in the code above.

closed 2024-11-20 by Marooned-MB

Marooned-MB, 2024-11-21:
If anyone will be doing those changes for this printer, I may suggest a few more that worked for me:
1. setting bigger font size ([line 256](https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod/blob/main/thumbnail.py#L256))
```Python
font_size = int(img_size.height() / 14);
->
font_size = int(img_size.height() / 10);
```
2. moving height to 2nd line not to overlap with time ([line 282](https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod/blob/main/thumbnail.py#L282)) - added `\n` and `AlignLeft`
```Python
draw_text(painter, rect_top, '\n' + self.max_height_formatted, Qt.AlignmentFlag.AlignLeft | Qt.AlignmentFlag.AlignTop, color, bgcolor)
```
3. adding `--short_duration_format` parameter in Orca for the next point
4. adjusting time format function ([lines 178..192](https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod/blob/main/thumbnail.py#L178)) - current function has a bug which results in double `d` for longer time, eg: `1d 2h 34m 56s` -> `1dd 02:34`
My version takes up to 3 most significant parts:
```
1d 2h 34m 56s => 1d 2h 34m
2h 34m 56s    => 2h 34m 56s
   34m 56s    => 34m 56s
```
->
```Python
            if self.print_duration_short_format:
                parts = re.findall(r'\s*(\d+)\s*([dhms])', self.print_duration)
                s = ' '.join(map(''.join, [i for i in parts[:3]])) # not sure if there is a nicer syntax
                self.print_duration_formatted = '\u29D6' + s
```

Here are a few photos - sorry for poor quality - phone refused to focus on the screen for some reasons:

Top two are the final version: `font_size / 10`, two lines on top, new time format
[image]

And here is the view of particular file - short and long one:
[image]
[image]

Marooned-MB, 2025-06-12:
@alkoka
Looks like the file has been dramatically changed since I've proposed the change. Either you referr to the older version (from the date I've posted the solution) or one would have to prepare a new change.
Because older version is working for me, I don't intend to prepare a patch for the new version at this moment, sorry :)

alkoka, 2025-06-12:
**That made it work! Thanks a lot!!!! Very happy now!**

So on the version from February 4 2024 (2.0.0)

the following has to be changed:

            header += self.image_encode_new(img_200x200, ';gimage:')
            header += self.image_encode_new(self.image_modify(self.image_resize(img, 160, 160)), ';simage:')

to the one below:

        header += self.image_encode_new(self.image_modify(self.image_resize(img, 85, 85)), ';simage:')
        header += self.image_encode_new(self.image_modify(self.image_resize(img, 230, 230)), ';gimage:')
        header += self.image_encode_new(self.image_modify(self.image_resize(img, 170, 170)), ';mimage:').

(Yes, two lines overwritten by three lines pasted.

Gcode thumbnails is set to 230x230.

Marooned-MB, 2025-06-12:
Glad I could help ^_^

arantius, 2026-03-14:
I'm gathering enough details that I'll post a new issue to write it all out.  Short version: I'm seeing "long lines" because sometimes this tool seems to use `\r` and sometimes `\n` as a newline delimiter, but at least my text editor doesn't like the bare carriage returns.

(Also: thanks for being so responsive!)