fifonik/ElegooNeptuneThumbnailPrusaMod #16: Thumbnails view for Elegoo Neptune 3 PLUS on OrcaSlicer

TalbotJ2412, 2024-12-12:
I installed OrcaSlicer V2.2.0
I want to add thumbnails view for my Neptune 3 PLUS print jobs.
Followed instructions from https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod

Repository is located in C:\3D_PrintSW\ElegooNeptuneThumbnailPrusaMod-2.0.0
I installed Python V3.13 from Microsoft Store.

I did printer settings in OrcaSlicer.
The 'Post-processing Scripts' section read like this:
--- --- ---
"C:\Users\jean_\AppData\Local\Microsoft\WindowsApps\python3.exe" 
"C:\3D_PrintSW\ElegooNeptuneThumbnailPrusaMod-2.0.0\thumbnail.py";
--- --- ---

I use the Retraction Cubes STL from MatterHackers.

Upon execution of the Export GCODE file, I see the console window opening for 2 seconds then closing.
Then I get this error message:
--- --- ---
Post-processing script
"C:\Users\jean_\AppData\Local\Microsoft\WindowsApps\python3.exe"
"C:\3D_PrintSW\ElegooNeptuneThumbnailPrusaMod-2.0.0\thumbnail.py"
on file
C:\Users\jean_\AppData\Local\Temp/orcaslicer_model/Wed_Dec_11/21_49_55#13892#57/Metadata\.13892.0.gcode.pp
failed.
Error code: 1
--- --- ---

Same error message with other versions of Python. (python.exe, pythonw.exe)

I installed 'pip install pyinstaller' to try to create the executable.
Got a warning, however install successfull:
--- --- ---
  WARNING: The scripts pyi-archive_viewer.exe, pyi-bindepend.exe, pyi-grab_version.exe, pyi-makespec.exe, pyi-set_version.exe and pyinstaller.exe are installed in 'C:\Users\jean_\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.13_qbz5n2kfra8p0\LocalCache\local-packages\Python313\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
--- --- ---
I tried to execute 'pyinstaller build.spec', but I don't know how to navigate to the repo folder in the console.
Didn't found out when looking at the Python docs. (First time I see Python code !)

There is no `build.bat` file in the repo.

I can't find a thumbnail.log file on my PC (Windows 10). Looks like it's not created.

Jean

fifonik, 2025-01-21:
Sorry, notifications were disabled for the repository so I've missed all these issues.

> I installed Python V3.13 from Microsoft Store.
> "C:\Users\jean_\AppData\Local\Microsoft\WindowsApps\python3.exe" "C:\3D_PrintSW\ElegooNeptuneThumbnailPrusaMod-2.0.0\thumbnail.py";

Have you installed Pythin into ur used directory?
By default Python installed in C:\PythinXYZ\...
Have you checked that you actually have executable `C:\Users\jean_\AppData\Local\Microsoft\WindowsApps\python3.exe`?
Have you checked if `C:\3D_PrintSW\ElegooNeptuneThumbnailPrusaMod-2.0.0\thumbnail.log` was created?

> There is no build.bat file in the repo.

Oh, Will fix this.

fifonik, 2025-03-18:
Closing as there is no feedback from user