fifonik/ElegooNeptuneThumbnailPrusaMod #18: Feature Request: Show price info

TrapStoner, 2025-03-13:
I would like to display price info rather than used filament length it's more useful to me and probably to others too. Similarly to the other scripts this one was built upon it would be sweet if you have price info display. It could be a simple argument toggle to display price info instead of the length for example.

fifonik, 2025-03-13:
I've made some changes and so now it is possible to configure what information will be displayed in every corner.
For your situation you can specify: 
`--icons print_duration,model_height,filament_weight,filament_cost` -- to display duration and model height on top and filament weight and price at the bottom
`--currency €` -- to specify currency character
`--currency_suffix` -- to add currency suffix after value

P.S. Pushed .py for now, will build exe in a few days

TrapStoner, 2025-03-13:
Hmm i get this error after building it even after doing `pip install -r requirements.txt`

[image]

fifonik, 2025-03-17:
Sorry, that I provided wrong option name (`icons` instead of `info`)

>  Also i think you can remove --currency_suffix

Without the option (default) currency char is added before numeric value ($10), with this option -- after numeric value (10$)

closed 2025-03-18 by fifonik