fifonik/ElegooNeptuneThumbnailPrusaMod #19: script issue

YasinCetiner, 2025-04-07:
"D:\OrcaSlicer\ElegooNeptuneThumbnailPrusaMod-main\dist\thumbnail.exe" "--info" "print_duration,model_height,filament_weight,filament_cost" i want to see filament cost in screen but when i add this script, thumbnail not showing version is orca 2.3 without extra arguments thumbnail works.

fifonik, 2025-04-07:
Are there any error message?
Can you provide thumbnail.log?
Can you try to add "--debug" then run and provide thumbnail.log?

YasinCetiner, 2025-04-08:
2025-04-08 07:59:44,066 - ERROR - Error occurred while running application.
Traceback (most recent call last):
  File "thumbnail.py", line 546, in <module>
  File "argparse.py", line 1823, in parse_args
  File "argparse.py", line 2575, in error
  File "argparse.py", line 2545, in print_usage
  File "argparse.py", line 2556, in _print_message
AttributeError: 'NoneType' object has no attribute 'write'

fifonik, 2025-04-08:
To be honest -- I do not know as I was not able to re-produce the issue. 

Could you provide screenshot of how thumbnail generation configured in OrcaSlicer. I'm talking about this one:
[image]

You can also try the following:
1. Download attached archive and extract file cube.gcode to D:\OrcaSlicer\ElegooNeptuneThumbnailPrusaMod-main\dist\cube.gcode
2. Open command prompt and navigate to the folder:
```
D:\
cd D:\OrcaSlicer\ElegooNeptuneThumbnailPrusaMod-main\dist
```
3. Ececute the following comand (copy paste it):
```
thumbnail.exe cube.gcode --debug --info print_duration,model_height,filament_weight,filament_cost
```
4. Run `dir` comand and check results

In my case after executing the command I see two png images and log file were generated aside of executable and as cube.gcode increased in size:
[image]

[cube.zip](https://github.com/user-attachments/files/19646037/cube.zip)

YasinCetiner, 2025-04-08:
[image] i changed this area and i can see filament cost on 3d printer. works without arguments on orca post script. i think orca 2.3 broke something on post script. klipper ui just shows this
[image] i did your said but get same error log and no png file. i can see everthing on printer but not klipper ui. just small issue

fifonik, 2025-04-08:
I'm using the script with OrcaSlicer 2.3.0 at this moment and do not have the issue (I'm using .py script, not .exe. but after your report I double tested with .exe and it works fine).
By default the script not updating original image displayed by fluidd (klipper UI). To do so you need to use option `--update_original_image`.

The piece of code you provided looks strange. In the most recent version there is no such code on line 278. In fact, I rid of "variables" print_duration_formatted, max_height_formatted etc some time ago and they are not in the current version. Are you sure that you using the most recent version?

Unfortunately, I cannot suggest anything else as you not providing details I asked so no any other ideas comes in my mind (screenshot from Orca's config, from cmd would help a lot, but you not providing it for some reason).

YasinCetiner, 2025-04-08:
i downloaded new version and everything works perfect. that's my fault sorry. thank u so much

closed 2025-04-08 by fifonik