fifonik/ElegooNeptuneThumbnailPrusaMod #22: It Does not work in bambu studio because the thumbnail size and format options are missing even though bambu studio is just orca slicer

Breadmasterpro, 2025-09-29:
[image]

fifonik, 2025-09-29:
Are there any images in gcode generated by Bambu Studio?
Could you slice a cube and supply gcode here please?

Breadmasterpro, 2025-09-30:
Here is another small gcode for good mesure.
[Body1_0.16mm_PLA_Elegoo Neptune 4 Pro_4m10s.zip](https://github.com/user-attachments/files/22609262/Body1_0.16mm_PLA_Elegoo.Neptune.4.Pro_4m10s.zip)

fifonik, 2025-09-30:
Well, actually I see a thumbnail in the gcode files you provided:

```
; THUMBNAIL_BLOCK_START
; thumbnail begin 50x50 648
; iVBORw0KGgoAAAANSUhEUgAAADIAAAAyCAYAAAAeP4ixAAABrUlEQVR4Ae2Yy2oqQRBAe6cufOALwf
; f7ragLRRD8tPmz+bS6noEKTTY3CSSpCXWgsLtnFnW6qhuZEBzHcRzHcZxcUiqVnvV6PeGXCHmj0Wgk
; y+VSNpuNbLdbWa/XslgspNvtpsVi8RnyQLPZTBDY7/dyPB7lcrnI4XDIhJCbTqdSq9WSYBkqgcRut8
; skrter3O/3bMyaVmYymUilUkmCRTgDJMrOU43T6SSPx0Nut9ubCJJalcFgIMEivV4vfS9CRc7nc9Za
; schsNpPhcGizKv1+PxOJzwcySDDX1opFXvL2qkLvr1art5uKxFWAOet6RmgtRF7ytkS4UufzebbbsY
; yGSvCc92KRQqHwDFZAhHaJZUhcgznrWo3RaJRJ0FrmRLhSVYaESVyDuVaC97ixkDApMh6PhSBZhFRK
; x6zzXA+5aRFahlApdp9QAa3E66+KfRES1qTjMWcCAY1ciWjowXaRn8JFXOSbeC8SS7jIb/BnWgu+Kh
; Ks4SLW4BPPZ0VarVYaLNLpdNKP3lpmJZRqtZr8T6RcLichL3CtItVut1N2n08/uRJwHMdxnDzyD0tw
; 9v24H1T1AAAAAElFTkSuQmCC
; thumbnail end
; THUMBNAIL_BLOCK_END
```

It is not in a format that supported by Neptunes and the script is not detecting it as there is no extension information.
However, I cat try to make some changes in the script so it will try to use the thumbnail.
Note, the in-built thumbnail is tiny (50x50 px) so the preview itself will have poor quality. In any case it will probably be better than nothing.

fifonik, 2025-09-30:
Well, actually the script might work without any modifications. Try option `--image_size 50x50`

fifonik, 2025-09-30:
You do not need to modify ur printer firmware. You will need to add script into your Bamboo Studio (I hope you can do that, in OrcaSlicer it is on "Other" tab). After that everything you slice in the slicer and send to printer will have these thumbnails.

P.S. I do not really understand why with Neptune you using Bambu Studio instead or OrcaSlicer. From understanding, the Bambu Studio have some features specific for their printers and other printers do not have the benefits anyway.

Breadmasterpro, 2025-09-30:
> Using Bambu Studio for non-Bambu printer while owning more than 1 printer makes perfect sense. Ta. Could you try to setup the script in Bambu Studio with image_size option specified (as above) and check if this work at all (steps 5 & 6 from the github main page)?

I am really sorry but I have no idea how to do this and google is not much help.. what am i doing wrong?

[image]

fifonik, 2025-10-01:
As per installation steps, it should be something like:
```
"C:\Path\Where\You\Put\thumbnail.exe" "--image_size=50x50";
```

Breadmasterpro, 2025-10-07:
> As per installation steps, it should be something like:
> 
> ```
> "C:\Path\Where\You\Put\Downloaded\thumbnail.exe" "--image_size=50x50";
> ```

thankyou